openthread/ot-esp32 #3: Infinite reboot on ESP32

beriberikix, 2020-05-23:
I've got everything flashed and wired but the ESP32 build keeps on rebooting. I have some differences in my setup than the instructions:

- I'm on macOS Catalina
- Using an older [devkit v2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards-previous.html#esp32-core-board-v2-esp32-devkitc)
- using the latest esp-idf, which is 4.x

When I run `make monitor` I can see this output before a reboot:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7004
load:0x40078000,len:14332
ho 0 tail 12 room 4
load:0x40080400,len:3980
entry 0x40080680
I (31) boot: ESP-IDF v4.2-dev-1660-g7d7521367 2nd stage bootloader
I (31) boot: compile time 12:31:38
I (31) boot: chip revision: 0
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot:  3 ot_storage       Unknown data     01 81 00110000 00080000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x104ec ( 66796) map
I (134) esp_image: segment 1: paddr=0x00020514 vaddr=0x3ffb0000 size=0x02354 (  9044) load
I (137) esp_image: segment 2: paddr=0x00022870 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/jonathanberi/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (141) esp_image: segment 3: paddr=0x00022c78 vaddr=0x40080400 size=0x0b194 ( 45460) load
I (169) esp_image: segment 4: paddr=0x0002de14 vaddr=0x00000000 size=0x02204 (  8708) 
I (172) esp_image: segment 5: paddr=0x00030020 vaddr=0x400d0020 size=0x4d1dc (315868) map
0x400d0020: _stext at ??:?

I (301) boot: Loaded app from partition at offset 0x10000
I (301) boot: Disabling RNG early entropy source...
I (301) cpu_start: Pro cpu up.
I (305) cpu_start: Application information:
I (310) cpu_start: Project name:     ot-cli
I (314) cpu_start: App version:      5b26882
I (319) cpu_start: Compile time:     May 23 2020 12:31:33
I (326) cpu_start: ELF file SHA256:  f3887eb3958ce710...
I (331) cpu_start: ESP-IDF:          v4.2-dev-1660-g7d7521367
I (338) cpu_start: Starting app cpu, entry point is 0x40081228
0x40081228: call_start_cpu1 at /Users/jonathanberi/esp/esp-idf/components/esp32/cpu_start.c:286

I (329) cpu_start: App cpu up.
I (349) heap_init: Initializing. RAM available for dynamic allocation:
I (355) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (361) heap_init: At 3FFB40F8 len 0002BF08 (175 KiB): DRAM
I (368) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (374) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (380) heap_init: At 4008B594 len 00014A6C (82 KiB): IRAM
I (387) cpu_start: Pro cpu start user code
I (405) spi_flash: detected chip: gd
I (406) spi_flash: flash io: dio
bort() was called at PC 0x400d36c7 on core 0
0x400d36c7: syscall_not_implemented_aborts at /Users/jonathanberi/esp/esp-idf/components/newlib/syscalls.c:39

Backtrace:0x400852df:0x3ffba4e0 0x4008597d:0x3ffba500 0x400891ce:0x3ffba520 0x400d36c7:0x3ffba590 0x40119da7:0x3ffba5b0 0x400e1549:0x3ffba5d0 0x400e1edc:0x3ffba5f0 0x400e1fac:0x3ffba610 0x400e0def:0x3ffba630 0x400d35a5:0x3ffba650 0x400872d1:0x3ffba6a0
0x400852df: panic_abort at /Users/jonathanberi/esp/esp-idf/components/esp_system/panic.c:330

0x4008597d: esp_system_abort at /Users/jonathanberi/esp/esp-idf/components/esp_system/system_api.c:100

0x400891ce: abort at /Users/jonathanberi/esp/esp-idf/components/newlib/abort.c:46

0x400d36c7: syscall_not_implemented_aborts at /Users/jonathanberi/esp/esp-idf/components/newlib/syscalls.c:39

0x40119da7: exit at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x400e1549: ot::Spinel::RadioSpinel<ot::Esp32::HdlcInterface, otSysMainloopContext>::WaitResponse() at /Users/jonathanberi/esp/ot-esp32/third_party/openthread/src/lib/spinel/radio_spinel_impl.hpp:1305 (discriminator 1)

0x400e1edc: ot::Spinel::RadioSpinel<ot::Esp32::HdlcInterface, otSysMainloopContext>::Init(bool, bool) at /Users/jonathanberi/esp/ot-esp32/third_party/openthread/src/lib/spinel/radio_spinel_impl.hpp:210

0x400e1fac: platformRadioInit at /Users/jonathanberi/esp/ot-esp32/src/radio.cpp:365

0x400e0def: otSysInit at /Users/jonathanberi/esp/ot-esp32/src/system.c:59

0x400d35a5: run_cli at /Users/jonathanberi/esp/ot-esp32/example/main/cli.c:59

0x400872d1: vPortTaskWrapper at /Users/jonathanberi/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: f3887eb3958ce710

Rebooting...
ets Jun  8 2016 00:22:57
```

I think the panic may either be a bug due to using 4.x or possible differences in the ESP32 module used.

Any suggestions?

beriberikix, 2020-05-23:
Switching to [v3.3.2](https://github.com/espressif/esp-idf/releases/tag/v3.3.2) made no difference:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6808
load:0x40078000,len:12072
load:0x40080400,len:6708
entry 0x40080778
I (71) boot: Chip Revision: 0
I (32) boot: ESP-IDF v3.3.2 2nd stage bootloader
I (32) boot: compile time 13:48:15
I (33) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot:  3 ot_storage       Unknown data     01 81 00110000 00080000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x14bc0 ( 84928) map
I (133) esp_image: segment 1: paddr=0x00024be8 vaddr=0x3ffb0000 size=0x01edc (  7900) load
I (137) esp_image: segment 2: paddr=0x00026acc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/freertos/xtensa_vectors.S:1779

I (140) esp_image: segment 3: paddr=0x00026ed4 vaddr=0x40080400 size=0x0913c ( 37180) load
I (164) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x4d3e4 (316388) map
0x400d0018: _stext at ??:?

I (276) esp_image: segment 5: paddr=0x0007d404 vaddr=0x4008953c size=0x01114 (  4372) load
0x4008953c: prvSwitchTimerLists at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/freertos/timers.c:859

I (284) boot: Loaded app from partition at offset 0x10000
I (284) boot: Disabling RNG early entropy source...
I (285) cpu_start: Pro cpu up.
I (289) cpu_start: Application information:
I (294) cpu_start: Project name:     ot-cli
I (298) cpu_start: App version:      5b26882
I (303) cpu_start: Compile time:     May 23 2020 13:48:13
I (309) cpu_start: ELF file SHA256:  9cc21ae594d23648...
I (315) cpu_start: ESP-IDF:          v3.3.2
I (320) cpu_start: Starting app cpu, entry point is 0x40081120
0x40081120: call_start_cpu1 at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/esp32/cpu_start.c:268

I (311) cpu_start: App cpu up.
I (331) heap_init: Initializing. RAM available for dynamic allocation:
I (338) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (344) heap_init: At 3FFB4470 len 0002BB90 (174 KiB): DRAM
I (350) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (356) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (363) heap_init: At 4008A650 len 000159B0 (86 KiB): IRAM
I (369) cpu_start: Pro cpu start user code
I (52) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
abort() was called at PC 0x400e0d83 on core 0
0x400e0d83: _exit at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/newlib/syscalls.c:95

ELF file SHA256: 9cc21ae594d23648a6abfc45f206821f387f48b1df5c144f31ac09019290947a

Backtrace: 0x40085e08:0x3ffba430 0x40086051:0x3ffba450 0x400e0d83:0x3ffba470 0x4011b247:0x3ffba490 0x400e1829:0x3ffba4b0 0x400e21d3:0x3ffba4d0 0x400e22ac:0x3ffba4f0 0x400e10b3:0x3ffba510 0x400d3553:0x3ffba530 0x40087372:0x3ffba580
0x40085e08: invoke_abort at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/esp32/panic.c:715

0x40086051: abort at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/esp32/panic.c:715

0x400e0d83: _exit at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/newlib/syscalls.c:95

0x4011b247: exit at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/exit.c:70

0x400e1829: ot::Spinel::RadioSpinel<ot::Esp32::HdlcInterface, otSysMainloopContext>::WaitResponse() at /Users/jonathanberi/esp/ot-esp32/src/radio.cpp:236

0x400e21d3: ot::Spinel::RadioSpinel<ot::Esp32::HdlcInterface, otSysMainloopContext>::Init(bool, bool) at /Users/jonathanberi/esp/ot-esp32/third_party/openthread/src/lib/spinel/radio_spinel_impl.hpp:210

0x400e22ac: platformRadioInit at /Users/jonathanberi/esp/ot-esp32/src/radio.cpp:365

0x400e10b3: otSysInit at /Users/jonathanberi/esp/ot-esp32/src/system.c:59

0x400d3553: run_cli at /Users/jonathanberi/esp/ot-esp32/example/main/cli.c:59

0x40087372: vPortTaskWrapper at /Users/jonathanberi/esp/esp-idf-v3.3.2/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57
```

wgtdkp, 2020-05-24:
@beriberikix The stack trace shows that the UART connection is not working. Please make sure that:
1. the `ot-rcp` firmware is built by following [this guide](https://github.com/openthread/ot-esp32/blob/master/example/README.md) (especially, you need to specify the serial number of your nrf52840 board with the `-s` option).
2. the ESP32 and nrf52840 boards are connected with the same computer.

beriberikix, 2020-05-24:
Ah, I was wondering if they needed a common ground. During my test the nRF was connected to a battery power source. I'll try a different configuration and report back.

beriberikix, 2020-05-24:
Adding a common ground fixed the issue! I'll suggest an addition to the README to close out this issue.

closed 2020-05-26 by jwhui

bunower, 2021-06-17:
Hello,
I have the same problem.

I'm using a nRF52840 MDK USB Dongle plugged in to the notebook and a AZDelivery ESP32 NodeMCU also plugged in to the notebook. 

Here's the bootloop message:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6832
ho 0 tail 12 room 4
load:0x40078000,len:14704
load:0x40080400,len:4292
entry 0x400806e4
I (76) boot: Chip Revision: 1
I (76) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0.2-583-g577aef29a 2nd stage bootloader
I (41) boot: compile time 20:27:35
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 ot_storage       Unknown data     01 81 00110000 00080000
I (100) boot: End of partition table
I (105) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x105a8 ( 66984) map
I (145) esp_image: segment 1: paddr=0x000205d0 vaddr=0x3ffb0000 size=0x0217c (  8572) load
I (148) esp_image: segment 2: paddr=0x00022754 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /root/esp-idf/components/freertos/xtensa_vectors.S:1778

I (152) esp_image: segment 3: paddr=0x00022b5c vaddr=0x40080400 size=0x0b498 ( 46232) load
I (180) esp_image: segment 4: paddr=0x0002dffc vaddr=0x00000000 size=0x02014 (  8212)
I (183) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x4e400 (320512) map
0x400d0018: _stext at ??:?

I (307) boot: Loaded app from partition at offset 0x10000
I (307) boot: Disabling RNG early entropy source...
I (307) cpu_start: Pro cpu up.
I (311) cpu_start: Application information:
I (316) cpu_start: Project name:     ot-cli
I (321) cpu_start: App version:      39a4915
I (326) cpu_start: Compile time:     Jun 17 2021 20:27:32
I (332) cpu_start: ELF file SHA256:  d55afeb9622b6661...
I (338) cpu_start: ESP-IDF:          v4.0.2-583-g577aef29a
I (344) cpu_start: Starting app cpu, entry point is 0x400813bc
0x400813bc: call_start_cpu1 at /root/esp-idf/components/esp32/cpu_start.c:272

I (330) cpu_start: App cpu up.
I (354) heap_init: Initializing. RAM available for dynamic allocation:
I (361) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (367) heap_init: At 3FFB4708 len 0002B8F8 (174 KiB): DRAM
I (374) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (380) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (386) heap_init: At 4008B898 len 00014768 (81 KiB): IRAM
I (392) cpu_start: Pro cpu start user code
I (411) spi_flash: detected chip: generic
I (411) spi_flash: flash io: dio
I (412) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
abort() was called at PC 0x400d3917 on core 0
0x400d3917: _exit at /root/esp-idf/components/newlib/syscalls.c:67

ELF file SHA256: d55afeb9622b6661

Backtrace: 0x400849b9:0x3ffba6c0 0x40084d3d:0x3ffba6e0 0x400d3917:0x3ffba700 0x4011b8a7:0x3ffba720 0x400e18dd:0x3ffba740 0x400e2268:0x3ffba760 0x400e2338:0x3ffba780 0x400e0e6b:0x3ffba7a0 0x400d357d:0x3ffba7c0 0x400871f5:0x3ffba810
0x400849b9: invoke_abort at /root/esp-idf/components/esp32/panic.c:155

0x40084d3d: abort at /root/esp-idf/components/esp32/panic.c:172

0x400d3917: _exit at /root/esp-idf/components/newlib/syscalls.c:67

0x4011b8a7: exit at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x400e18dd: ot::Spinel::RadioSpinel<ot::Esp32::HdlcInterface, otSysMainloopContext>::WaitResponse() at /root/ot-esp32/third_party/openthread/src/lib/spinel/radio_spinel_impl.hpp:1349 (discriminator 1)

0x400e2268: ot::Spinel::RadioSpinel<ot::Esp32::HdlcInterface, otSysMainloopContext>::Init(bool, bool) at /root/ot-esp32/third_party/openthread/src/lib/spinel/radio_spinel_impl.hpp:217

0x400e2338: platformRadioInit at /root/ot-esp32/src/radio.cpp:365

0x400e0e6b: otSysInit at /root/ot-esp32/src/system.c:60

0x400d357d: run_cli at /root/ot-esp32/example/main/cli.c:59

0x400871f5: vPortTaskWrapper at /root/esp-idf/components/freertos/port.c:143
```

I've tried also to connect the GND GPIOs between each other but no success. Could someone help me?

Thanks in advance & kind Regards

jwhui, 2021-06-18:
@bunower , Espressif recently added OpenThread support directly into their IDF. You may want to check out it: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli

PR #4 (2020-05-26): [docs] add note about common ground
Closes #3