openthread/ot-esp32 #5: [log] no return in log output

lanyuwen, 2020-06-08:
Log output looks messy due to lack of "returns" in log output. By default configuration, ESP32 will convert "\n" as "\r\n", however, the configuration seems having been reset during the ot-esp32's own UART configuration.

wgtdkp, 2020-06-08:
Yes, this is a known issue. We cannot adopt the default ESP32 configuration because UART is also used for communication with RCP (Unfortunately, ESP32 do not support configuring return-line per UART).

We provide a workaround in [example/README.md](https://github.com/openthread/ot-esp32/tree/master/example):
> Note: Dependending on your Linux distribution, the new line characters ('\r', '\n') may cause commands display issues. To have good look of the CLI, Connect the ESP32 board with advanced command picocom -b 115200 /dev/ttyUSB0 --imap lfcrlf (install picocom with sudo apt-get install picocom).

Please let me know if this works for you.

lanyuwen, 2020-06-09:
This works for me. However, this issue would impact all applications built upon ot-esp32 (e.g., Weave/CHIP border router). And we are not able to tell everywhere that people need to use a different terminal tool for ot-esp32.

Some thoughts:
1. we need to append '\r' for all logs of ot-esp32 itself since we are already aware of the problem
2. reach out to Espressif to see if there is a way to configure each UART separately? @chshu :)
3. possible to override "ESP_LOG" functions to append '\r'?

wgtdkp, 2020-06-09:
1. Yes, will do.
2. Since it will be a new feature, it may be only added to HEAD of esp-idf (while we and weave is using v3.3).
3. esp-idf has a definition of `LOG_FORMAT` as below:
    ```c++
    #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%d) %s: " format LOG_RESET_COLOR "\n"
    ```
    A platform independent implementation should append "\r\n" (OpenThread appends "\r\n" for logs) at the tail. Treating it as a bug, we may fix it in `esp-idf/v3.3`.

lanyuwen, 2020-06-09:
Sounds good!

>    A platform independent implementation should append "\r\n" (OpenThread appends "\r\n" for logs) at the tail. Treating it as a bug, we may fix it in `esp-idf/v3.3`.

Will you submit a PR? This sounds like a decent fix to me, though not sure if they are willing to accept it as ESP IDF will append '\r' automatically by default.

chshu, 2020-06-09:
@koobest for visibility

Synced with the driver team, they prefer to add a new API to config the line-endings per UART, so there is no side effect to the existing users, the feature could be merged to master first and cherry-pick to v3.3.

As a short term workaround, you can hardcode something here:
https://github.com/espressif/esp-idf/blob/master/components/vfs/vfs_uart.c#L1011 

s_ctx[OT_RADIO_UART_NUM]->tx_mode = ESP_LINE_ENDINGS_LF;
s_ctx[OT_CLI_UART_NUM]->tx_mode = ESP_LINE_ENDINGS_CRLF;

wgtdkp, 2020-06-09:
@chshu thanks for replying!
```C++
s_ctx[OT_RADIO_UART_NUM]->tx_mode = ESP_LINE_ENDINGS_LF;
s_ctx[OT_CLI_UART_NUM]->tx_mode = ESP_LINE_ENDINGS_CRLF;
```
This works for our case. For long term solution, instead of appending `"\r\n"`, configuring the line-endings per UART will also work prefect!

wgtdkp, 2020-06-09:
@chshu 
oops! It seems that esp-idf/v3.3 and esp-idf/master is different at here:
https://github.com/espressif/esp-idf/blob/d7e6eecd66c63b5d4f725060603bd13b43b29b3c/components/vfs/vfs_uart.c#L928-L931
```c++
void esp_vfs_dev_uart_set_tx_line_endings(esp_line_endings_t mode)
{
    s_tx_mode = mode;
}
```
And we cannot make this workaround with even master branch, because s_ctx is defined as `static` while type `vfs_uart_context_t` is not exposed at all (I thought it should work at first glance but failed to compile the workaround).

chshu, 2020-06-09:
The team is working on this now. 

Yes, the workaround meant to modify the code in vsf_uart.c.

wgtdkp, 2020-06-09:
@chshu Thanks to you and the esp-idf team!
I made another workaround here https://github.com/openthread/ot-esp32/pull/9, would you like to take a look if the fix should work?

chshu, 2020-06-09:
Looks like a workable workaround to me:)

wgtdkp, 2020-06-09:
> Looks like a workable workaround to me:)

@chshu Thanks for confirmation!

lanyuwen, 2020-06-09:
> The team is working on this now.
> 
> Yes, the workaround meant to modify the code in vsf_uart.c.

Thank you Shu! 👍 👍 @chshu

closed 2020-06-12 by jwhui