openthread/ot-esp32 #6: [cli] scan command without thread configuration will crash the system

lanyuwen, 2020-06-08:
scan command without thread configuration will crash the system, no problem if using it after thread interface gets up.

wgtdkp, 2020-06-08:
This is a bug in OpenThread `RadioSpinel` implementation. PR  https://github.com/openthread/openthread/pull/5068 is submitted and you can checkout and test at your local.

I will later update the openthread submodule once the PR got merged.

jwhui, 2020-06-12:
https://github.com/openthread/openthread/pull/5068 has been merged.

wgtdkp, 2020-06-15:
> [openthread/openthread#5068](https://github.com/openthread/openthread/pull/5068) has been merged.

will fire a PR after testing on real device with latest openthread tomorrow.

lanyuwen, 2020-06-15:
it won't crash anymore, however, still no output until thread interface is up.

wgtdkp, 2020-06-16:
@lanyuwen 
> it won't crash anymore, however, still no output until thread interface is up.

the `scan` command scans for existing Thread Networks, is there already an Thread Network that is reachable from the ESP32 node?

lanyuwen, 2020-06-16:
I can get some results after issuing "ifconfig up", or "ifconfig up" and "ifconfig down". However, I got nothing unless these commands executed.

I don't think this behavior is expected. But I think this is a low priority problem and can be tracked by a separated issue. Let's close this one for now.

lanyuwen, 2020-06-17:
reopened to track the scan failure fix progress

closed 2020-06-17 by jwhui