openthread/ot-esp32 #13: List of ESP32 devices that are 802.15.4-capable

drobertson123, 2020-07-14:
I can find no indication on the Espressif web site that ESP32 is 802.15.4 capable. Is there a resource somewhere that shows what chips will work and which will not?

I would love to use this library but I don't know if/where it will function.

jwhui, 2020-07-14:
I'm not aware of any existing ESP32 devices that include IEEE 802.15.4. This repo currently requires an IEEE 802.15.4 SoC that implements the [Radio Co-Processor](https://openthread.io/platforms#radio-co-processor-rcp). OpenThread on the ESP32 will communicate with the RCP via UART.

Hope that helps.

drobertson123, 2020-07-14:
Thank you. That clears up a significant point of confusion.

wgtdkp, 2020-07-15:
> I would love to use this library but I don't know if/where it will function.

@drobertson123 you may start with the the [example](https://github.com/openthread/ot-esp32/tree/master/example) application.

bunower, 2021-06-01:
What about
Espressif ESP32-WROOM-32D ?
is this capable of IEEE 802.15.4

gabekassel, 2021-08-05:
https://www.espressif.com/en/news/ESP32_H2 🤞

unitware, 2021-11-25:
@gabekassel that will do 802.15.4, but not WiFi :(

jwhui, 2021-11-25:
> @gabekassel that will do 802.15.4, but not WiFi :(

@unitware, Espressif supports a multi-chip Wi-Fi + 802.15.4 solution. See the [OpenThread Border Router Guide](https://openthread.io/guides/border-router/espressif-esp32) for more details.

unitware, 2021-11-25:
Yes, but I was looking forward to a single cpu solution and was misreading the press releases in the beginning 

Skickat från min iPhone

jwhui, 2022-03-25:
Closing stale issue.

closed 2022-03-25 by jwhui