openthread/ot-esp32 #17: Support for ESP8266

NotClear, 2020-11-22:
Does it work with an esp8266?

wgtdkp, 2020-11-22:
> Does it work with an esp8266?

We didn't do any verification on esp8266. This project requires two UARTs (one for RCP and one for command line) to work. Since both esp8266 and esp32 has double UARTs, I think there should be hardware limitations stop it from working on esp8266. Though you may need to do some migration.

jwhui, 2020-11-22:
> I think there should be hardware limitations stop it from working on esp8266 (though you may need to do some migration).

@wgtdkp , did you mean that there should be **_no_** hardware limitations?

wgtdkp, 2020-11-22:
> > I think there should be hardware limitations stop it from working on esp8266 (though you may need to do some migration).
> 
> @wgtdkp , did you mean that there should be **_no_** hardware limitations?

@jwhui my typo, thanks for pointing it out.

NotClear, 2020-11-22:
> > Does it work with an esp8266?
> 
> We didn't do any verification on esp8266.
> 
> This project requires two UARTs (one for RCP and one for command line) to work. Since both esp8266 and esp32 has double UARTs, I think there should be **no** hardware limitations stop it from working on esp8266 (though you may need to do some migration).

are there plans to migrate this software in the future? many ioT devices already have an esp8266 installed.

NotClear, 2020-11-22:
i am new to the thread topic so i am sorry for my simple questions. 

Does the ESP32 run on its own with the software or is an additional transmitter and receiver necessary for the integration and use of the ESP32 like for example a "nRF52840"? @wgtdkp @jwhui

jwhui, 2020-11-22:
> Does the ESP32 run on its own with the software or is an additional transmitter and receiver necessary for the integration and use of the ESP32 like for example a "nRF52840"? @wgtdkp @jwhui 

@NotClear , Thread requires an IEEE 802.15.4 radio. An ESP32 alone is not sufficient to support Thread and must be paired with an IEEE 802.15.4 radio.

wgtdkp, 2020-11-24:
@NotClear you may take a look at the [example](https://github.com/openthread/ot-esp32/blob/master/example/README.md) to learn the setup and requirements.

closed 2020-12-14 by jwhui