openthread/ot-esp32 #18: can not add on-mesh prefix

cahty82008, 2021-04-29:
I'm using the same board as in the tutorial, esp32-WROOM-32+ nrf52840DK

I followed tutorial and got the borad running. 

Then i connected to the ESP32 use "make monitor" command, and tried to set up a border router based on this example and added a prefix, i expected to see the new IP address based on the prefix, but with no luck, here is the log:

```
> dataset init new
Done
> dataset panid 0xdead
Done
> dataset extpanid dead1111dead2222
Done
> dataset networkname OpenThreadGuide
Done
> dataset masterkey 11112233445566778899DEAD1111DEAD
Done
> dataset pskc 198886f519a8fd7c981fee95d72f4ba7
Done
> dataset commit active
I (109910) OT_ESP32_PLAT: [INFO]-MESH-CP-: Active dataset set
Done
> I (109910) OT_ESP32_PLAT: [INFO]-CORE----: Notifier: StateChanged (0x00004000) [Channel] 
prefix add fd11:22::/64 pasor
Done
>   ifconfig up
Done
> I (131240) OT_ESP32_PLAT: [INFO]-CORE----: Notifier: StateChanged (0x01001009) [Ip6+ LLAddr Ip6Mult+ NetifState] 
thread start
I (142060) OT_ESP32_PLAT: [NOTE]-MLE-----: Role Disabled -> Detached
I (142060) OT_ESP32_PLAT: [INFO]-MLE-----: Attempt to become router
I (142080) OT_ESP32_PLAT: [INFO]-CORE----: Non-volatile: Read NetworkInfo {rloc:0xe400, extaddr:16acb86936cbbde8, role:Leader, mode:0x0f, version:2, keyseq:0x0, ...
I (142090) OT_ESP32_PLAT: [INFO]-CORE----: Non-volatile: ... pid:0x7579b8e1, mlecntr:0xbc3, maccntr:0xbb8, mliid:7a58456ff467c9ef}
I (142100) OT_ESP32_PLAT: [INFO]-CORE----: Non-volatile: Saved NetworkInfo {rloc:0xe400, extaddr:16acb86936cbbde8, role:Leader, mode:0x0f, version:2, keyseq:0x0, ...
I (142120) OT_ESP32_PLAT: [INFO]-CORE----: Non-volatile: ... pid:0x7579b8e1, mlecntr:0xfac, maccntr:0xfa0, mliid:7a58456ff467c9ef}
I (142130) OT_ESP32_PLAT: [INFO]-MLE-----: Send Link Request (ff02:0:0:0:0:0:0:2)
Done
> I (142130) OT_ESP32_PLAT: [INFO]-CORE----: Notifier: StateChanged (0x00001035) [Ip6+ Role MLAddr Rloc+ Ip6Mult+] 
I (142160) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:82, chksum:6208, to:0xffff, sec:no, prio:net
I (142170) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (142180) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
I (154463) OT_ESP32_PLAT: [NOTE]-MLE-----: RLOC16 e400 -> fffe
I (154473) OT_ESP32_PLAT: [INFO]-MLE-----: AttachState Idle -> Start
I (154473) OT_ESP32_PLAT: [INFO]-CORE----: Notifier: StateChanged (0x00000040) [Rloc-] 
I (154973) OT_ESP32_PLAT: [NOTE]-MLE-----: Attempt to attach - attempt 1, any-partition reattaching with Active Dataset
I (154973) OT_ESP32_PLAT: [INFO]-MLE-----: AttachState Start -> ParentReqRouters
I (154993) OT_ESP32_PLAT: [INFO]-MLE-----: Send Parent Request to routers (ff02:0:0:0:0:0:0:2)
I (155023) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:84, chksum:c0a7, to:0xffff, sec:no, prio:net
I (155023) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (155033) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
I (155753) OT_ESP32_PLAT: [INFO]-MLE-----: AttachState ParentReqRouters -> ParentReqReeds
I (155753) OT_ESP32_PLAT: [INFO]-MLE-----: Send Parent Request to routers and REEDs (ff02:0:0:0:0:0:0:2)
I (155783) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:84, chksum:7aa9, to:0xffff, sec:no, prio:net
I (155793) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (155803) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
I (157023) OT_ESP32_PLAT: [INFO]-MLE-----: AttachState ParentReqReeds -> Idle
I (157023) OT_ESP32_PLAT: [NOTE]-MLE-----: Allocate router id 57
I (157043) OT_ESP32_PLAT: [NOTE]-MLE-----: RLOC16 fffe -> e400
I (157043) OT_ESP32_PLAT: [NOTE]-MLE-----: Role Detached -> Leader
I (157043) OT_ESP32_PLAT: [NOTE]-MLE-----: Leader partition id 0x29f12921
I (157063) OT_ESP32_PLAT: [INFO]-CORE----: Notifier: StateChanged (0x000012a5) [Ip6+ Role Rloc+ PartitionId NetData Ip6Mult+] 
I (157073) OT_ESP32_PLAT: [INFO]-N-DATA--: Sent server data notification
I (157073) OT_ESP32_PLAT: [INFO]-MLE-----: Send Data Response (ff02:0:0:0:0:0:0:1)
I (157093) OT_ESP32_PLAT: [INFO]-CORE----: Non-volatile: Saved NetworkInfo {rloc:0xe400, extaddr:16acb86936cbbde8, role:Leader, mode:0x0f, version:2, keyseq:0x0, ...
I (157103) OT_ESP32_PLAT: [INFO]-CORE----: Non-volatile: ... pid:0x29f12921, mlecntr:0xfaf, maccntr:0xfa0, mliid:7a58456ff467c9ef}
I (157113) OT_ESP32_PLAT: [INFO]-N-DATA--: Received network data registration
I (157113) OT_ESP32_PLAT: [INFO]-N-DATA--: Allocated Context ID = 1
I (157133) OT_ESP32_PLAT: [INFO]-N-DATA--: Sent network data registration acknowledgment
I (157133) OT_ESP32_PLAT: [INFO]-CORE----: Notifier: StateChanged (0x00000200) [NetData] 
I (157143) OT_ESP32_PLAT: [INFO]-MLE-----: Send Data Response (ff02:0:0:0:0:0:0:1)
I (157173) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:96, chksum:1ca4, to:0xffff, sec:no, prio:net
I (157173) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (157183) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
I (157213) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:118, chksum:036f, to:0xffff, sec:no, prio:net
I (157223) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (157233) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
I (157633) OT_ESP32_PLAT: [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
I (157653) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:0972, to:0xffff, sec:no, prio:net
I (157663) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (157673) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
I (159813) OT_ESP32_PLAT: [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
I (159833) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:62b5, to:0xffff, sec:no, prio:net
I (159843) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (159853) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
I (162553) OT_ESP32_PLAT: [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
I (162583) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:0773, to:0xffff, sec:no, prio:net
I (162583) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (162593) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
netdataregister
Done
> I (168203) OT_ESP32_PLAT: [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
I (168223) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:d2ec, to:0xffff, sec:no, prio:net
I (168223) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (168233) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
ipaddr
**fd10:bd85:fa1a:a351:0:ff:fe00:fc00
fd10:bd85:fa1a:a351:0:ff:fe00:e400
fd10:bd85:fa1a:a351:7a58:456f:f467:c9ef
fe80:0:0:0:14ac:b869:36cb:bde8**
Done
> I (180323) OT_ESP32_PLAT: [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
I (180343) OT_ESP32_PLAT: [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:1814, to:0xffff, sec:no, prio:net
I (180353) OT_ESP32_PLAT: [INFO]-MAC-----:     src:[fe80:0:0:0:14ac:b869:36cb:bde8]:19788
I (180363) OT_ESP32_PLAT: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
> prefix
fd11:22:0:0::/64 paros med
Done

```

wgtdkp, 2021-04-29:
Hi,

You need to enable the OPENTHREAD_CONFIG_IP6_SLAAC_ENABLE feature to have
the SLAAC address of the new prefix.
I submitted https://github.com/openthread/ot-esp32/pull/19 for this issue,
please let me know if it doesn't work.

FYI, there is already an openthread example in the espressif/esp-idf:
https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli
and this project will be deprecated in the future.

BRs,
Kangping

cahty82008, 2021-04-29:
Hi @wgtdkp , 
thank you for the very quick reply and the information about the example, i will check this example.
enable the **OPENTHREAD_CONFIG_IP6_SLAAC_ENABLE** works,
but compile the [#19 ]  throw the following errors:

```
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp: In member function 'otError ot::MeshCoP::Dtls::Setup(bool)':
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp:291:5: error: 'mbedtls_ssl_conf_handshake_timeout' was not declared in this scope
     mbedtls_ssl_conf_handshake_timeout(&mConf, 8000, 60000);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp:291:5: note: suggested alternative: 'mbedtls_ssl_conf_read_timeout'
     mbedtls_ssl_conf_handshake_timeout(&mConf, 8000, 60000);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     mbedtls_ssl_conf_read_timeout
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp:300:9: error: 'mbedtls_ssl_conf_dtls_cookies' was not declared in this scope
         mbedtls_ssl_conf_dtls_cookies(&mConf, mbedtls_ssl_cookie_write, mbedtls_ssl_cookie_check, &mCookieCtx);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp:300:9: note: suggested alternative: 'mbedtls_ssl_conf_sig_hashes'
         mbedtls_ssl_conf_dtls_cookies(&mConf, mbedtls_ssl_cookie_write, mbedtls_ssl_cookie_check, &mCookieCtx);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         mbedtls_ssl_conf_sig_hashes
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp: In member function 'otError ot::MeshCoP::Dtls::SetClientId(const uint8_t*, uint8_t)':
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp:528:16: error: 'mbedtls_ssl_set_client_transport_id' was not declared in this scope
     int rval = mbedtls_ssl_set_client_transport_id(&mSsl, aClientId, aLength);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ot-esp32/third_party/openthread/src/core/meshcop/dtls.cpp:528:16: note: suggested alternative: 'mbedtls_ssl_conf_transport'
     int rval = mbedtls_ssl_set_client_transport_id(&mSsl, aClientId, aLength);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                mbedtls_ssl_conf_transport
/home/esp-idf/make/component_wrapper.mk:291: recipe for target 'third_party/openthread/src/core/meshcop/dtls.o' failed
make[1]: *** [third_party/openthread/src/core/meshcop/dtls.o] Error 1
/home/esp-idf/make/project.mk:590: recipe for target 'component-ot-esp32-build' failed
make: *** [component-ot-esp32-build] Error 2
```

wgtdkp, 2021-04-29:
I pushed the latest fix, could you try the PR again?

cahty82008, 2021-04-29:
yeah, I can compile it successfully now, the prefix also works.

I have a question, how could I access the outside network ?
wpantund is for Unix-like operating systems, do you have similar tools or examples for ESP32 to route packets between IPv6 and the thread network?

wgtdkp, 2021-04-29:
> do you have similar tools or examples for ESP32 to route packets between IPv6 and the thread network?

The short answer is no and I don't think there is plan of further development of this repo to support that. I would suggest you create an issue in espress-if/esp-idf repo to ask for a Thread Border Router support.

cahty82008, 2021-04-29:
ok, thank you for the reply and the suggestion

PR #19 (2021-04-30): [config] enable SLAAC
Fixes #18
closed 2021-04-30 by jwhui