openthread/ot-esp32 #20: Not able to compile for thunderboard sense2

vineet2888, 2021-07-14:
Problem : Not able to compile for thunderboard sense2(silicon lab) BOARD="BRD4166A".

Enviroment : 
1)Prepared setup based on documentation https://github.com/openthread/ot-esp32/blob/master/example/README.md.
2)compilation for esp32 is successfull 
3)compilation for thunderboard sense 2 is failed.
4)used "make -f examples/Makefile-efr32mg12 CFLAGS="-DUART_HWFC_ENABLED=0"  command to compile binary for thunderboard sense2

Error log:
(reverse-i-search)`': ^C
[root@mx():/home/vineet/openthread/ot-esp32/third_party/openthread#] make -f examples/Makefile-efr32mg12 CFLAGS="-DUART_HWFC_ENABLED=0"
  BUILD    efr32mg12
make -j10 -C build/efr32mg12 --no-print-directory all
  CHECK         .local-version
make --no-print-directory all-recursive
Making all in include
make  all-recursive
Making all in openthread
Making all in platform
make[6]: Nothing to be done for 'all'.
make[6]: Nothing to be done for 'all-am'.
Making all in third_party
Making all in silabs
make[4]: *** No rule to make target 'gecko_sdk_suite/v2.7/hardware/kit/common/bsp/bsp_bcc.c', needed by 'gecko_sdk_suite/v2.7/hardware/kit/common/bsp/libsilabs_efr32mg12_sdk_a-bsp_bcc.o'.  Stop.
make[4]: *** Waiting for unfinished jobs....
  CC       rail_config/libsilabs_efr32mg12_sdk_a-rail_config.o
/home/vineet/openthread/ot-esp32/third_party/openthread/examples/../third_party/silabs/rail_config/rail_config.c:16:10: fatal error: em_common.h: No such file or directory
 #include "em_common.h"
          ^~~~~~~~~~~~~
compilation terminated.
make[4]: *** [Makefile:1576: rail_config/libsilabs_efr32mg12_sdk_a-rail_config.o] Error 1
make[3]: *** [Makefile:483: all-recursive] Error 1
make[2]: *** [Makefile:591: all-recursive] Error 1
make[1]: *** [Makefile:517: all] Error 2
make: *** [examples/Makefile-efr32mg12:293: do-build-cortex-m4] Error 2
[root@mx():/home/vineet/openthread/ot-esp32/third_party/openthread#]

vineet2888, 2021-07-14:
Moreover not able to find 'gecko_sdk_suite in the current project. 

Please let me know if i am missing any step .

jwhui, 2021-07-14:
@vineet2888 , a couple notes:

1. Espressif now includes OpenThread support in their IDF - I encourage you to leverage that https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli

1. If you really want to use this repo, then refer to [third_party/openthread/repo/examples/platforms/efr32mg12/README.md](https://github.com/openthread/openthread/blob/3b362071e594375035e063c0c25988b1aa52bd5f/examples/platforms/efr32mg12/README.md) for instructions on how to build.

vineet2888, 2021-07-19:
Thanks @jwhui for the reply.

When i am following the suggested method 1. i am getting following core dump for ot_cli and ot_br

x4012657b:0x3ffd1530 0x4008c071:0x3ffd15c0
0x400d38c3: panic_abort at /home/pi/openthread/esp-idf/components/esp_system/panic.c:379

0x40088d61: esp_system_abort at /home/pi/openthread/esp-idf/components/esp_system/esp_system.c:126

0x4008f7f2: abort at /home/pi/openthread/esp-idf/components/newlib/abort.c:46

0x401260e6: syscall_not_implemented_aborts at /home/pi/openthread/esp-idf/components/newlib/syscalls.c:34

0x40179d17: exit at /builds/idf/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x400f1eda: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::HandleRcpTimeout() at /home/pi/openthread/esp-idf/examples/openthread/ot_br/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:2190

0x400f1f2c: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::WaitResponse() at /home/pi/openthread/esp-idf/examples/openthread/ot_br/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:1662

0x400f24a8: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::Init(bool, bool, bool) at /home/pi/openthread/esp-idf/examples/openthread/ot_br/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:242

0x400f265d: esp_openthread_radio_init at /home/pi/openthread/esp-idf/components/openthread/port/esp_openthread_radio_uart.cpp:35

0x4010b525: esp_openthread_platform_init at /home/pi/openthread/esp-idf/components/openthread/port/esp_openthread_platform.cpp:58

0x4012fff1: esp_openthread_init at /home/pi/openthread/esp-idf/components/openthread/port/esp_openthread.cpp:40

0x4012657b: ot_task_worker at /home/pi/openthread/esp-idf/examples/openthread/ot_br/build/../main/esp_ot_br.c:155

0x4008c071: vPortTaskWrapper at /home/pi/openthread/esp-idf/components/freertos/port/xtensa/port.c:168

I am using silicon lab geeko wifi starter kit flashed with ot_rcp and esp32 as host controller.

jwhui, 2021-07-19:
@vineet2888 , for ESP-IDF issues, I would do one or both of the following:

1. Post an issue at https://github.com/espressif/esp-idf/issues
2. Reach out to Espressif directly: https://www.espressif.com/en/contact-us/technical-inquiries

jwhui, 2022-03-25:
Closing stale issue.

closed 2022-03-25 by jwhui