openthread/ot-esp32 #22: Bootstrap script fails, 'bc' and 'clang-format-6.0' not found

ianmtaylor1, 2024-06-19:
I'm following the instructions found on the [Build OpenThread Guide](https://openthread.io/guides/build/index.md) using the openthread/environment docker container, and I ran into the following two errors. When running `./script/bootstrap`, I get:

```
third_party/openthread/script/bootstrap: 48: bc: not found
third_party/openthread/script/bootstrap: 48: [: Illegal number:
```
and
```
E: Unable to locate package clang-format-6.0
E: Couldn't find any package by glob 'clang-format-6.0'
E: Couldn't find any package by regex 'clang-format-6.0'
```
The first error is resolved by running `sudo apt install bc` beforehand, so this should either be done in the script or bc should be included in the openthread/environment docker container.

The second error is less straightforward. It appears to be generated from line 60, `sudo apt-get --no-install-recommends install -y clang-format-6.0`. clang-format-14 is installed already in the docker container. Will this version be sufficient? If so, perhaps this installation could be removed from the script.

jwhui, 2024-06-20:
@ianmtaylor1 , thanks for raising this issue.

This repo hasn't been maintained since 2021. I suggest taking a look at this codelab for the latest regarding OpenThread on ESP32: https://openthread.io/codelabs/esp-openthread-hardware

/cc/ @chshu

ianmtaylor1, 2024-06-20:
Thanks! I will look there. I think this issue can also be closed, then.